rugix/rugix #4: support older boards without the `tryboot` feature

koehlma, 2023-11-20:
In principle, it should be possible to add support for older boards without the `tryboot` feature by using [U-Boot](https://docs.u-boot.org/en/latest/) as an intermediate bootloader. U-Boot would then be responsible for switching between A/B partitions. This would also open up the possibility to use Rugpi with other boards than Raspberry Pi. If you want to see this happen, please give this issue a 👍.

## Known Caveats

- There can only be a single `config.txt` and it needs to be configured to load the bootloader instead of the kernel.

## Open Questions

- [ ] Consider using [RAUC](https://rauc.readthedocs.io/en/latest/index.html) as a basis and implement the current `tryboot` switching within RAUC.

reubenmiller, 2023-11-20:
[swupdate](https://sbabic.github.io/swupdate/overview.html) might also be worth evaluating as it does have the potential to stream the new images to partition (though I haven't used it personally).

Streaming the new image directly to the cold partition is attractive for devices with little available disk space. Mender supports streaming by default.

koehlma, 2023-11-24:
I decided against using RAUC or SWUpdate as a basis (for now). Integrating with them would add a lot of complexity (both in terms of required configuration and the building pipeline). As it turns out, getting a U-Boot setup working was easier than anticipated. And, by doing this ourselves, we gain more flexibility going ahead and overall achieve greater consistency between the different boot flows.

Rugpi Bakery does now support an option `boot_flow = "u-boot"` to enable the U-Boot boot flow (this also disables the `tryboot` boot flow). I modeled the U-Boot boot flow after the `tryboot` boot flow such that they are more consistent (still need to update the docs). In particular, committing works by writing a file to the `config` partition.

I tested the image built with the U-Boot boot flow and the Bullseye-based release of Raspberry Pi OS on a Raspberry Pi 3. In principle, the same image should work for Zero 2W as the device tree loaded by the Raspberry Pi firmware is used (or at least that is what the U-Boot documentation claims).

@reubenmiller Could you please test, whether it works on the Zero 2W?

koehlma, 2023-11-27:
Rugpi now also has experimental support for streaming images directly to the SD card by using `--stream`. You can use

```sh
rugpi-ctrl update install --stream -
```

to install an image streamed to stdin, e.g., via `curl`. This also allows using compressed images by piping them through `xz` (or `gzip`). Mid-term streaming should also become the default.

reubenmiller, 2023-11-27:
@koehlma I must be doing something wrong, as I can't get the image to boot on Raspberry PI 3 or Zero 2W.

I've set the boot flow in the `rugpi-bakery.toml` (see my project here: https://github.com/reubenmiller/rpi-tedge-image/tree/support-rp3), and I've made sure the latest docker image

```toml
boot_flow = "u-boot"
```

I've tried the following images (buster and bookworm):

* [2023-10-10-raspios-bookworm-arm64-lite.img.xz](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-10-10/2023-10-10-raspios-bookworm-arm64-lite.img.xz)
* [2023-05-03-raspios-bullseye-arm64-lite.img.xz](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-05-03/2023-05-03-raspios-bullseye-arm64-lite.img.xz)

koehlma, 2023-11-27:
No, you are doing everything right. GitHub Actions does not include LFS files by default. The U-Boot binaries are stored in LFS. Instead of them, their stubs were then baked into the image. Commit 94fd6e7 should fix that. If you are building the Docker image yourself, please make sure that you have a working LFS setup.

reubenmiller, 2023-11-27:
Perfect. I’ll give the images a try again in the next day.

reubenmiller, 2023-11-28:
I've been able to successfully test initial bootstrapping and using the u-boot and everything works as expected 🥇 

**Procedure**

1. Build an image then flash the SD Card (using the `boot_flow = "u-boot"` option)
2. Boot device for the first time with flashed SD Card
3. Perform an update (using the same) image, going from A -> B partition

**Devices under Test**

* ✅ Raspberry Pi Zero 2 W Rev 1.0 (CPU: BCM2835)
* ✅ Raspberry Pi 3 Model B Rev 1.2 (CPU: BCM2835)

✅ = PASS

reubenmiller, 2023-11-28:
Any idea on the effort for Raspberry Pi 1, 2 and Zero? I would be happy to assist with testing (if that helps).

koehlma, 2023-11-28:
That's great news!

For 32-bit support, first Rugpi Bakery needs to be made aware of the architecture (currently, it simply assumes `arm64`). As far as I know, there are three different architectures which we would have to support: `arm64` (ARMv8), `armhf` (ARMv7), and `armel` (ARMv6). At least this is how the CPU families fit into Debian's architecture schema.

Raspberry Pi 2 B v1.2 is actually based on ARMv8 (same chip family as Pi 3) and should support 64-bit. However, I am not sure whether this is properly supported by the bootrom (first stage bootloader burned into the ROM of the chip). I can test the existing U-Boot boot flow with it later – maybe it just works. ;)

In any case, supporting `armhf`, i.e., the entire Raspberry Pi 2 family, should be rather easy. We would have to produce U-Boot binaries and adapt Rugpi Bakery to place them together with an appropriate `config.txt` in the image. Rugpi Ctrl gets build for `armhf` already. The [recipe installing it](https://github.com/silitics/rugpi/blob/main/recipes/rugpi-ctrl/steps/01-run.sh) simply needs to be made aware of the architecture. This should easily be doable in an afternoon.

For Raspberry Pi 1 and Zero, I am not sure. Actually, it is still a bit unclear to me how there can just be one `armhf` image apparently working on all Pis. It seems like this involves some customization as it falls between the officially supported Debian architectures `armhf` and `armel`.[^1] I guess, building Rugpi Ctrl for `arm-unknown-linux-musleabihf` instead of `armv7-unknown-linux-musleabihf` would do the trick – but I have to investigate this further.

Do you think that someone starting a new project with Rugpi today would like to use those older boards? Or, is this just so that existing setups can switch to Rugpi? I would like to understand the use case better before investing time into this.

[^1]: https://wiki.debian.org/RaspberryPi

reubenmiller, 2023-11-28:
> That's great news!
> 
> For 32-bit support, first Rugpi Bakery needs to be made aware of the architecture (currently, it simply assumes `arm64`). As far as I know, there are three different architectures which we would have to support: `arm64` (ARMv8), `armhf` (ARMv7), and `armel` (ARMv6). At least this is how the CPU families fit into Debian's architecture schema.
> 
> Raspberry Pi 2 B v1.2 is actually based on ARMv8 (same chip family as Pi 3) and should support 64-bit. However, I am not sure whether this is properly supported by the bootrom (first stage bootloader burned into the ROM of the chip). ~I can test the existing U-Boot boot flow with it later – maybe it just works. ;)~ I tried using the existing image and U-Boot binary and copied over the respective `.dtb`s but it did not work.
> 
> In any case, supporting `armhf`, i.e., the entire Raspberry Pi 2 family, should be rather easy. We would have to produce U-Boot binaries and adapt Rugpi Bakery to place them together with an appropriate `config.txt` in the image. Rugpi Ctrl gets build for `armhf` already. The [recipe installing it](https://github.com/silitics/rugpi/blob/main/recipes/rugpi-ctrl/steps/01-run.sh) simply needs to be made aware of the architecture. This should all be easily doable in an afternoon.
> 
> For Raspberry Pi 1 and Zero, I am not sure. Actually, it is still a bit unclear to me how there can just be one `armhf` image apparently working on all Pis. It seems like this involves some customization as it falls between the officially supported Debian architectures `armhf` and `armel`.[1](#user-content-fn-1-c407769ac990a54bfdcfc03a359b3c1f) I guess, building Rugpi Ctrl for `arm-unknown-linux-musleabihf` instead of `armv7-unknown-linux-musleabihf` would do the trick – but I have to investigate this further.
> 
> Do you think that someone starting a new project with Rugpi today would like to use those older boards? Or, is this just so that existing setups can switch to Rugpi? I would like to understand the use case better before investing time into this.
> 
> ## Footnotes
> 1. https://wiki.debian.org/RaspberryPi [↩](#user-content-fnref-1-c407769ac990a54bfdcfc03a359b3c1f)

I was able to run the same image and do an image update on a `Raspberry Pi 2 Model B Rev 1.2`. I've done one successful update, so I'm happy to monitor future updates on this device to see if it could fall under the "supported" category.

Generally I think you can skip support for Raspberry 1 and Zero as these devices are very limited (only single core processors). I was just curious to see if it was an easy-win or not, as some customers are pushing for lower cost devices all the time and consider using older devices with lower specs just to save the per-unit cost....however I would still avoid those customers to avoid a single core cpu if possible.

koehlma, 2023-11-29:
There may now be support for Pi 1, Pi 2, and Pi Zero. You need to use the following settings:

```
architecture = "armhf"
boot_flow = "u-boot"
```

They have to be used for `customize` and for `bake`.

And, of course, a 32-bit base image must be used (Bookworm worked for me).

I did test the resulting image on a Pi 1 B+ v1.2 and it seems to work.

@reubenmiller Can you please test whether this works on Pi Zero and Pi 2? The image should be identical for all those boards as the bootloader is selected by a filter in `config.txt`.

Took me a bit less than an hour to get that working.

koehlma, 2023-12-03:
Great, thanks!

Indeed, the whole situation with the different architectures is a bit of a mess. In general, the architecture of a binary is automatically determined by `binfmt_misc` and the binary is then run via QEMU. We are not doing anything special here. I guess, this is then also what `uname` reports. Rugpi Ctrl is also compiled for `arm-unknown-linux-musleabihf`. Unfortunately, I do not think that we can do anything about it, except mentioning it in the docs.

I will close this issue now. It was an easy win after all. ;)

I am now also confident that we can rather easily support other boards, should the need arise.

As the next step, I will focus on introducing a layer architecture similar to Docker and enabling delta updates that way.