rugix/rugix #8: idea: Option to enable set-persist by default in an image

reubenmiller, 2023-11-23:
Currently the official way to enable `rugpi-ctrl state overlay set-persist true`, however there does not seem to be an option to enable the setting by default during the build process.

The current workaround is to create a recipe that installs a one-shot service which enables the setting on first boot (and using the systemd `first-boot-complete.target` option)

**Use Case**

In a perfect world, `set-persist true` would not be required as everything the user would need would be part of the image. Or if it did not included everything, at least the directories/files that should be parked as persisted should be known at build time.

In practice, it is very difficult/impossible to known all of the directories/files that should be persisted at build time. Therefore using the `set-persist true` can be attractive because it allows users to customize their images to their liking (e.g. install custom applications, custom configuration files etc.)

* Install custom applications / delta updates
* Allow persistence of files without the user having to know much about the underlying persistence layer

Though it must be noted that if the user then upgrades their image and switches partition (e.g. A -> B), then any user-created state in the current partition will be lost unless if they backup/restore data from a persistent folder. However this is trivial to support (and the default in cases like mender), where the `/data` partition is a persisted across A/B partitions, therefore it enables user the ability to backup and restore their files if necessary (assuming they have tooling around the calling of the install new image command)

**Proposed Solution**

Add option in the `rugpi-bakery.toml` to enable `set-persist true` by default.

```toml
default_persist = true
```

koehlma, 2023-11-24:
Did you try using the `first-boot-complete.target`? As far as I know, Systemd detects the first boot by a missing (or uninitialized) `/etc/machine-id`. Rugpi initializes this file before handing over to Systemd, so, I think that the first boot mechanism of Systemd will never trigger. This was also intended to be a feature, as it is unclear what “first boot” means: First boot after provisioning, first boot after an update, or first boot after factory reset… Hence, I rather have this consistently disabled, than it triggering on some arbitrary definition of “first boot”. What are your thoughts on that?

Currently, Rugpi Ctrl is configured via recipes, e.g., by installing state files. I think, I would like to keep it that way. We can make this an option in a `rugpi-ctrl.toml` which will then be baked into the image by a recipe.

closed 2023-11-26 by koehlma

koehlma, 2023-11-26:
You can now enable persistency by default by installing a configuration file `/etc/rugpi/ctrl.toml` with:

```toml
overlay = "persist"
```

I also added a cautionary note to the documentation.

commit dd8df0ed (2023-11-26): feat: add option to make overlay persistent by default (closes #8)