rugix/rugix #9: Option to pass different rugpi-bakery.toml files when baking an image

reubenmiller, 2023-11-28:
Different Raspberry pi devices (pi 3, 4 and 5) requires slightly different configuration options (e.g. boot_flow, include_firmware etc.)

Currently rugpi-bakery expects that the image definition is defined in the `rugpi-bakery.toml` file, this does not allow to create different image variants which use the same recipes, but just differ slightly...To enable this you would have to have 1 image definition per repository which becomes more difficult to manage.

### Proposal

* rugpi-bakery supports providing the configuration file as an optional flag (defaulting to the existing value `rugpi-bakery.toml`)

**Example usage**

```sh
./run-bakery customize --config rugpi-baker.pi3.yaml
./run-bakery customize --config rugpi-baker.pi4.yaml
./run-bakery customize --config rugpi-baker.pi5.yaml
./run-bakery customize --config rugpi-baker.pi4.with_firmware.yaml
```

koehlma, 2023-11-28:
You can now use `--config` as you suggested. The flag must be in front of the command, e.g.:

```
./run-bakery --config rugpi-bakery.pi3.toml customize build/base.tar build/customized-pi3.tar
```

FYI: Technically, `boot_flow` and `include_firmware` only have an impact on the `bake` step. Hence, you could also use the default configuration to build `customized.tar` and then use a simple specific configuration like

```
boot_flow = "u-boot"
```

only for the `bake` step. Mid-term, I would like to support this officially. A possibility would be to have an `images` section in `rugpi-bakery.toml` where you could then declare the images you would like to `bake`. Before implementing something like that, I would like to think a bit more about the general design of the build pipeline, though. In particular, I think, it would also be nice to add some kind of layers which are cached and can be turned into independent artifacts for delta updates.

closed 2023-11-28 by koehlma

reubenmiller, 2023-11-28:
Perfect. Thanks for the quick work.

I'm currently working on the integration of rugpi with thin-edge.io...so hopefully I'll have something running soonish.