rugix/rugix #11: Support for signed updates

aqaengineering, 2023-12-05:
Hey there - this looks like a great project that could have a big impact on those building products on RPi. Thanks for putting it out there!

Wondering whether signed update support is on the roadmap at some point? Signed updates are a must-have for those using Rugpi to ship commercial products and may also be important for hobbyists as well.

Thanks again for the work on the project and making it open source with permissive license!

koehlma, 2023-12-05:
> this looks like a great project that could have a big impact on those building products on RPi

Thanks – I hope so. :)

> Wondering whether signed update support is on the roadmap at some point?

This is definitely something we have on the roadmap, however, it is not a priority right now. Currently, our focus is on solving the hard parts of OTA updates and state management and providing a solid basis. Note that you can already check signatures as part of a larger update process which involves Rugpi. For instance, if you provide a web form for installing updates, then you could verify the signature of the image before passing it on to Rugpi.

Something along the following lines should do the trick:

```shell
gpg --verify update.img.sig update.img
if [ $? == 0 ]; then
    rugpi-ctrl update install update.img
else
    echo "Signature verification failed!"
    exit 1
fi
```

Raspberry Pi also supports secure boot, which is also on our roadmap for the distant future as it would also allow encrypting the filesystem and making it almost completely tamperproof, thereby also protecting IP.

aqaengineering, 2023-12-06:
Makes sense, thanks for the quick reply and example code!

closed 2023-12-06 by aqaengineering