rugix/rugix #15: Images built based on Raspberry Pi OS with desktop cannot boot

kfihihc, 2024-02-05:
I referred to the [rugpi-template](https://github.com/silitics/rugpi-template) project and modified the `layers/customized.toml` to raspios-desktop (referenced from here: https://github.com/kfihihc/tedge-rugpi-core/tree/main/layers). I successfully built it using `sudo ./run-bakery bake image tryboot build/image-pi5-rugpi.img`. However, after flashing it to the SD card, it fails to boot. The official fan HAT of Raspberry Pi 5 keeps spinning, then stops, waits for a while, and then starts spinning again, indicating that there seems to be a problem with the boot process.

Additionally, when I referred to the rugpi-template project and built an image based on Raspberry Pi OS Lite, it could boot normally. Using `rugpi-ctrl update install image-pi5-rugpi.img` on a normally booted RPi OS Lite system to OTA to RPi OS Desktop results in an error message saying `No enough space or such file`.

Is there anything else I need to be aware of when building images based on RPi OS Desktop with Rugpi?

koehlma, 2024-02-05:
I never tried building an image on top of Raspberry Pi OS Desktop, to be honest.

Nevertheless, here are a few ideas on what the problem may be and how to fix it:

You need to make sure that `include_firmware = "none"` is set. In case you used the thin-edge.io template as a basis, I just submitted a PR earlier today fixing the issue: https://github.com/thin-edge/tedge-rugpi-image/pull/67. The official template has also been broken and erroneously included the firmware update for Pi 4 which will not work on Pi 5.

By default, the size of the system partition is 4 GB. In case of Raspberry Pi Desktop, this may be insufficient resulting in a broken system when bootstrapping (updates will simply fail, as you noticed). This is something, we need to fix in the future (see #16). For now, you need to increase the size manually in `/etc/rugpi/ctrl.toml`, e.g.:
```toml
system_size = "8G"
```
Note that this is incompatible with the thin-edge.io `persist-overlay` recipe which will overwrite the `ctrl.toml` configuration file to persist the overlay by default. If you also want that, use:
```toml
system_size = "8G"
overlay = "persist"
```

I hope that this resolves the issue.

kfihihc, 2024-02-06:
It's helpful! Thank you.

I cloned the latest rugpi-template and tested it with RPi OS Desktop. It works.

I also increased the system_size to 8G in /etc/rugpi/ctrl.img, and then it successfully performed OTA updates on another RPi OS Desktop.

However, it seems it cannot open some applications, like `chromium-browser `:

```bash
pi@rpi5-d83addeaaaa3:~ $ chromium-browser 
[4653:4653:0206/083040.393400:FATAL:zygote_host_impl_[linux.cc](http://linux.cc/)(127)] No usable sandbox! Update your kernel or see https://chromium.googlesource.com/chromium/src/+/main/docs/linux/suid_sandbox_development.md for more information on developing with the SUID sandbox. If you want to live dangerously and need an immediate workaround, you can try using --no-sandbox.
Trace/breakpoint trap
```

I'm wondering if it's because of the Read-only file system?

koehlma, 2024-02-06:
Perfect! 👍

The filesystem should be writeable as Rugpi mounts a writeable overlay over the original root filesystem. Just for clarification: Does Chromium work when installed on a normal installation of Raspberry Pi OS Desktop? Do other applications fail to start for the same/similar reason? Sandboxing seems very specific to Chromium. Did you install the [`chromium-sandbox` package](https://packages.debian.org/unstable/chromium-sandbox)?

closed 2024-03-11 by koehlma