rugix/rugix #17: Support for customizing /boot/config.txt

cmatheson, 2024-02-09:
I noticed that the some changes are made to the `config.txt` file during build. It would be nice if it were possible to add other options at some point as well (maybe some kind of ability to add lines the layer.toml?

(For now I'm just making the changes out-of-band which also works fine).

koehlma, 2024-02-09:
If you use the `tryboot` boot flow (Pi4/Pi5), you can use a recipe to copy your `config.txt` to `/boot`. Rugpi will simply remove any line starting with `auto_initramfs`, as this option is currently incompatible with Rugpi's own initialization procedure. Other than that, all customizations you make should be kept. Note that in the future the directory may change to `/boot/firmware` (which I consider to be a breaking change).

For `u-boot`, the `config.txt` can currently not be customized. I would have to give this some thought as the file cannot be updated and requires some specific options for U-Boot to work.

koehlma, 2024-03-11:
I gave this some thought. The only clean way to apply customizations to `config.txt` is when the `tryboot` boot flow is used. Otherwise, it is unclear what will happen when installing an update. Customizing `config.txt` is already supported by recipes modifying it in `/boot` directly. So, I see no need to add complexity to the build pipeline, e.g., in terms of layer configuration.

closed 2024-03-11 by koehlma