rugix/rugix #19: Baking pi4 image fails

raryanpur, 2024-02-15:
I'm working through the Getting Started documentation and encountered an error when building the rpi4 image via `./run-bakery bake image pi4 build/image-pi4.img`.

Running this on an M2 MacBook with the latest Docker Desktop (4.27.2 (137060)), which includes Docker Engine 25.0.3. Haven't changed any settings so both the Virtualization Framework and VirtioFS are enabled.

Build output is:

```
➜  rugpi-template git:(main) ./run-bakery bake image pi4 build/image-pi4.img
v0.6: Pulling from silitics/rugpi-bakery
Digest: sha256:de7e5b2aac92b093767f87cc797de91a25c9e0a26fe4b5256834d912ab68a65c
Status: Image is up to date for ghcr.io/silitics/rugpi-bakery:v0.6
 INFO baking image `pi4`
Cloning into '/project/.rugpi/repositories/fe0a146c8643a35e63802830c78f74e69ff196de'...
 INFO baking layer `customized`
 INFO baking layer `raspios-bookworm`
 INFO downloading `https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2023-12-11/2023-12-11-raspios-bookworm-arm64-lite.img.xz`
 INFO decompressing XZ image
 INFO creating `.tar` archive with system files
 INFO extracting system files
 INFO [ 1/14] apt: update package lists {}
 INFO     - 00-install.sh
 INFO [ 2/14] apt: upgrade all packages {}
 INFO     - 00-install.sh
 INFO [ 3/14] remove unnecessary Raspberry Pi OS functionality {}
 INFO     - 00-install.sh
Removed "/etc/systemd/system/multi-user.target.wants/userconfig.service".
resize2fs_once.service is not a native service, redirecting to systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable resize2fs_once
Failed to disable unit, unit unattended-upgrades.service does not exist.
Removed "/etc/systemd/system/multi-user.target.wants/sshswitch.service".
 INFO [ 4/14] disable swapping {}
 INFO     - 00-install.sh
 INFO [ 5/14] install and configure Rugpi Ctrl {"rugpi_admin": "true"}
 INFO     - 00-packages
 INFO     - 01-run.sh
 INFO     - 02-install.sh
	not a dynamic executable
Created symlink /etc/systemd/system/multi-user.target.wants/rugpi-admin.service → /lib/systemd/system/rugpi-admin.service.
 INFO [ 6/14] install state file for `/etc/fake-hwclock.data` {}
 INFO     - 00-install.sh
 INFO [ 7/14] install state file for `/root` {}
 INFO     - 00-install.sh
 INFO [ 8/14] sets the hostname {"hostname": "rugpi-template"}
 INFO     - 00-install.sh
 INFO [ 9/14] install Nginx {}
 INFO     - 00-packages
 INFO     - 01-install.sh
Synchronizing state of nginx.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable nginx
 INFO [10/14] enabling SSH {"root_authorized_keys": "ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABgQC2tN+ZL8lGnx8rW+vm9ukX1uhKE7GREHitIVxIN3fVh406ksaZzY4FB7JqMqor4SBpR/Eigm6mSSE6KdwSYYC99hakLVvFUG6b6xvB7gOgre8M0JuL9XwBfaUfNln6Hl2Xirlml61JwOWa8Lh+T8mquw9OUK20tkXNPrigVKH+RMQA2V0AQrHnzo9GXMT5HEdAfaVVhL8S1inlKixiPbnvt+nWUSoKGLo+I08w5ZKI88C+saP6VqFiinp57uF0F3oqwcupZe0j6vxGuN5hFg8YGcICFnjXUAVjds8pfcf7aImvYZdp192jC9JAfzx3LzJZLn+kY9hIQkqip/tfTtp56eBb+j9i07PhrDsGiZVNOWf+YG3Cw5Ix6ltOL0dvF1/xFG7O+CGz62w8Y925ytuGgzBkVJ090eznnCjpw+lhNiNFmoqUjiJFjs/VSrqmC5bqdRrqF7YIs61uKl/EyAZaEoHZJazUFFauOjjPK0ksVbAAfqxG4nFmOG0URemSvNE= koehlma@Zaylee\n"}
 INFO     - 00-install.sh
Synchronizing state of ssh.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable ssh
Created symlink /etc/systemd/system/sshd.service → /lib/systemd/system/ssh.service.
Created symlink /etc/systemd/system/multi-user.target.wants/ssh.service → /lib/systemd/system/ssh.service.
Removed "/etc/systemd/system/multi-user.target.wants/regenerate_ssh_host_keys.service".
Created symlink /etc/systemd/system/multi-user.target.wants/hydrate-ssh-host-keys.service → /lib/systemd/system/hydrate-ssh-host-keys.service.
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABgQC2tN+ZL8lGnx8rW+vm9ukX1uhKE7GREHitIVxIN3fVh406ksaZzY4FB7JqMqor4SBpR/Eigm6mSSE6KdwSYYC99hakLVvFUG6b6xvB7gOgre8M0JuL9XwBfaUfNln6Hl2Xirlml61JwOWa8Lh+T8mquw9OUK20tkXNPrigVKH+RMQA2V0AQrHnzo9GXMT5HEdAfaVVhL8S1inlKixiPbnvt+nWUSoKGLo+I08w5ZKI88C+saP6VqFiinp57uF0F3oqwcupZe0j6vxGuN5hFg8YGcICFnjXUAVjds8pfcf7aImvYZdp192jC9JAfzx3LzJZLn+kY9hIQkqip/tfTtp56eBb+j9i07PhrDsGiZVNOWf+YG3Cw5Ix6ltOL0dvF1/xFG7O+CGz62w8Y925ytuGgzBkVJ090eznnCjpw+lhNiNFmoqUjiJFjs/VSrqmC5bqdRrqF7YIs61uKl/EyAZaEoHZJazUFFauOjjPK0ksVbAAfqxG4nFmOG0URemSvNE= koehlma@Zaylee

 INFO [11/14] install and configure ZSH {"make_default": "true"}
 INFO     - 00-packages
 INFO     - 01-run.sh
--2024-02-15 21:47:23--  https://git.grml.org/f/grml-etc-core/etc/zsh/zshrc
Resolving git.grml.org (git.grml.org)... 202.61.209.103
Connecting to git.grml.org (git.grml.org)|202.61.209.103|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://git.grml.org/?p=grml-etc-core.git;a=blob_plain;f=etc/zsh/zshrc;hb=HEAD [following]
--2024-02-15 21:47:23--  https://git.grml.org/?p=grml-etc-core.git;a=blob_plain;f=etc/zsh/zshrc;hb=HEAD
Reusing existing connection to git.grml.org:443.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/plain]
Saving to: '/tmp/.tmpSE5c9b/etc/zsh/zshrc'

     0K .......... .......... .......... .......... ..........  256K
    50K .......... .......... .......... .......... ..........  524K
   100K .......... .......... .......                          35.5M=0.3s

2024-02-15 21:47:24 (435 KB/s) - '/tmp/.tmpSE5c9b/etc/zsh/zshrc' saved [130117]

 INFO     - 02-install.sh
 INFO [12/14] Raspberry Pi meta recipe {}
 INFO [13/14] install static webpage {}
 INFO     - 00-install.sh
 INFO [14/14] apt: cleanup cache and delete package lists {"autoremove": "true"}
 INFO     - 00-install.sh
 INFO packing system files
 INFO creating image (size: 2465550336 bytes)
fallocate: cannot open build/image-pi4.img: No such file or directory
Error: error running command `fallocate -l 2465550336 build/image-pi4.img`: command failed with non-zero exit code 1
=== STDERR ===
fallocate: cannot open build/image-pi4.img: No such file or directory
```

koehlma, 2024-02-15:
You have to create the _build_ directory first. I will add this to the documentation later (or patch Rugpi such that it creates the directory for you.

raryanpur, 2024-02-15:
Thanks, that did it!

commit a44ad4ed (2024-02-16): chore: create image parent directory (fixes #19)
closed 2024-02-15 by raryanpur