rugix/rugix #23: Generating and extracting a unique image ID

raryanpur, 2024-04-10:
Hi there, is there a recommended way to generate and extract a unique image ID from a built image? I'd like to build an image within a CI pipeline running in GitHub Actions. During the build, a step would create a random ID (i.e. using `tr` and `/dev/urandom` and write it to a file within the rootfs. Then, from within the GitHub Action the ID would be extracted from the image and saved to a db somewhere. 

On my machine I figure I could just mount the built image using Docker and extract the file. Or, just generate the ID file on my machine and copy it into the rootfs as part of the recipe. But I'm not sure how that would work with GitHub Actions. Figured I'd ask in case I'm missing something obvious. Thanks!

koehlma, 2024-04-10:
You can generate the ID with a recipe and then write it to a file in the project directory. The project directory is exposed to recipes via the `RUGPI_PROJECT_DIR` environment variable. So, something like

```shell
echo "${IMAGE_ID}" >"${RUGPI_PROJECT_DIR}IMAGE_ID"
```

should work. That way, you do not need to mount the image.

raryanpur, 2024-04-11:
Great, thank you!

closed 2024-04-11 by raryanpur