rugix/rugix #30: how to bake Raspberry OS Desktop image

dvorak0, 2024-07-16:
I got the same problem mentioned in https://github.com/silitics/rugpi/issues/15. 

I still find the baked image for Raspberry OS Desktop can not boot.

What I have done is:
1. install the latest version, which I believe is v0.7.4
2. use `./run-bakery init rpi-raspios` to get template files
3. enter `./run-bakery shell`, and directly change the url in `/usr/share/rugpi/repositories/core/layers/raspios-bookworm.arm64.toml`
4. run `./run-bakery image tryboot-pi4`

Then the image can not boot normally.

Thanks for your help.

koehlma, 2024-07-16:
Hey, thanks for reporting this issue.

The Docker container created by `./run-bakery` is ephemeral, i.e., any changes made with `./run-bakery shell` will not persist when invoking `./run-bakery` again. So, the image you built is likely just a regular Raspberry Pi OS Lite image. Can you please clarify what you mean by “can not boot normally”? Does that mean that the UI does not come up (which would be expected as the image is still based on Raspberry Pi OS Lite) or was there some error?

In general, Raspberry Pi OS Desktop is not out-of-the-box right now. That's also why there is no pre-defined `core` layer for it. That being said, it is definitely possible to build an image based on Raspberry Pi OS Desktop. You can copy the layer file (`raspios-bookworm.arm64.toml`) to your local project (in the `layers` directory) and change the URL there. You then need to change the `parent` layer of the `customized` layer that ships with the template to use your local copy with the changed URL. In addition, some manual tweaking of the core recipes, in particular, everything related to `core/rpi-raspios-setup` may be required to actually make it work.

dvorak0, 2024-07-17:
Thanks for you reply.

I know editing files inside shell is not persistent. So I run `./rugpi-bakery` inside the shell, without existing to host. From the size (~6.5GB compared 2.1GB) I think the image is indeed based on Desktop. That reminds me of a problem that will my 16GB card work for that size? I'll try another bigger card later.

The screen shows: 
[image]

dvorak0, 2024-07-17:
update: I've tried a much larger card, still cannot get it to work.

koehlma, 2024-07-17:
> I know editing files inside shell is not persistent. So I run `./rugpi-bakery` inside the shell, without existing to host.

I see. Yes, that should work.

The error message shows that the SD card could not be partitioned.

You need a larger SD card and you need to set a larger system size. How that is done is explained in the original issue you referred to: https://github.com/silitics/rugpi/issues/15#issuecomment-1927015729 (We should add this to the documentation.) Can you try whether it works when you set a larger system size (like 8 GB)?

I will need some time to investigate this further. There may be an edge case there that is triggered when the system partition does not match the configured size.

dvorak0, 2024-07-17:
> > I know editing files inside shell is not persistent. So I run `./rugpi-bakery` inside the shell, without existing to host.
> 
> I see. Yes, that should work.
> 
> The error message shows that the SD card could not be partitioned.
> 
> You need a larger SD card and you need to set a larger system size. How that is done is explained in the original issue you referred to: [#15 (comment)](https://github.com/silitics/rugpi/issues/15#issuecomment-1927015729) (We should add this to the documentation.) Can you try whether it works when you set a larger system size (like 8 GB)?
> 
> I will need some time to investigate this further. There may be an edge case there that is triggered when the system partition does not match the configured size.

Yes, I planned to set a larger system_size. Could you point for me where `/etc/rugpi/ctrl.toml` is?

dvorak0, 2024-07-17:
I will use `echo 'system_size = "8G" >> /etc/rugpi/ctrl.toml` in `01-install.sh` to try.

koehlma, 2024-07-17:
It has to be placed in the system being built. You need a custom recipe to do so.

dvorak0, 2024-07-17:
I confirm setting `system_size =6GB` works, even for my 16GB sdcard.

koehlma, 2024-07-17:
I would still recommend a larger SD card as otherwise the data partition is rather small.

dvorak0, 2024-07-17:
> I would still recommend a larger SD card as otherwise the data partition is rather small.

Sure. I've done this to double check the issue is round `system-size`. I think this issue can be closed.

dvorak0, 2024-07-22:
I have a following up question: is it allowed to upgrade from a version with smaller `system_size`, to a new version with larger `system_size`? If it is not allowed, that means I need to reserve a much larger` system_size` for future versions.

koehlma, 2024-07-22:
You can update as long as the actual update size fits into the system size used for the _initial_ installation. So, yes, you need to reserve as much space as you may need for future versions.

dvorak0, 2024-07-22:
Then I know otherwise the update will fail in the case of that the actual update exceeds the `system_size` used in the first time installation. Thanks for your suggestion.

closed 2024-08-29 by dvorak0