rugix/rugix #31: Failed to upgrade when changing base layer/image

dvorak0, 2024-09-03:
I tried to upgrade the device from version 1 to version 2. The difference between these two versions is that they use different base layers (`parent = version_1` and `parent = version_2`). Specifically, the version 2 has a newer version of linux kernel, and thus different boot partition content I assume.

The result I got is that the upgrading fails, and hot/spare slot switch never happens.

So could I know, is this kinds of upgrading allowed in current design?

Thanks.

koehlma, 2024-09-03:
Yes, such an update should be possible. Can you share any further details (e.g., an error message)?

dvorak0, 2024-09-03:
I got two problems:

1. Firstly I tried to build an image using [pi-gen](https://github.com/RPi-Distro/pi-gen). I confirmed the new image works. But when I start to bake rugpi image based on it. I got the following error.

>  INFO [ 3/15] remove unnecessary Raspberry Pi OS functionality {"disable_swapfile": "true"}
 INFO     - 00-install.sh
 INFO Mounting "/usr/share/rugpi/repositories/core/recipes/rpi-raspios-cleanup" to "/tmp/.tmpsyLdg0/roots/system/run/rugpi/bakery/recipe".
 INFO Mounting "/tmp/.tmpsyLdg0" to "/tmp/.tmpsyLdg0/roots/system/run/rugpi/bakery/bundle/".
resize2fs_once.service is not a native service, redirecting to systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable resize2fs_once
Failed to disable unit, unit unattended-upgrades.service does not exist.
Removed "/etc/systemd/system/multi-user.target.wants/sshswitch.service".
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dos2unix pi-language-support
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  piwiz* userconf-pi*
0 upgraded, 0 newly installed, 2 to remove and 1 not upgraded.
After this operation, 487 kB disk space will be freed.
(Reading database ... 132337 files and directories currently installed.)
Removing piwiz (0.75) ...
Removing userconf-pi (0.9) ...
(Reading database ... 132308 files and directories currently installed.)
Purging configuration files for userconf-pi (0.9) ...
/run/rugpi/bakery/recipe/steps/00-install.sh: line 16:  1034 Segmentation fault      (core dumped) apt-get purge -y userconf-pi
Error: error running command `chroot /tmp/.tmpsyLdg0/roots/system /run/rugpi/bakery/recipe/steps/00-install.sh`: command failed with non-zero exit code 139

2. I previously tried to install the kernel deb file using rugpi receipes. But I failed because the boot partition is not writable, which I think is intended behavior.

May I know how should I proceed?

Thanks

koehlma, 2024-09-03:
I do not know what goes wrong there. The error seems to indicate a segmentation fault in `apt-get` which is really strange, to say the least. In principle, you should be able to install a kernel with a recipe. Could you share a few more details about your setup? What architecture are your building for and what is your host's architecture and OS? What exactly did you try to install the kernel (if possible, please share the recipe you used) and what was the error message? Which version of Rugpi are you using and what `target` did you configure? There are no limitations regarding the boot partition (at least with the `tryboot` boot flow) – you can update anything on it.

dvorak0, 2024-09-04:
I can reproduce the problem: added kernel installation script in [this](https://github.com/dvorak0/bug-rugpi/blob/6ef7b60d70df1656f68bb640183de7c546e48084/recipes/hello-world/steps/01-install.sh#L19). The generated image is bootable, but the `/boot/config.txt` doesn't contain the line of `kernel=kernel8.img`.

koehlma, 2024-09-04:
I see. Yes, indeed, the boot partition is not mounted at `/boot`.

I haven't tested it, but I think the following should work:

```
BOOT_DIR="${RUGPI_BUNDLE_DIR}/roots/boot"
cp /boot/kernel8.img "${BOOT_DIR}"
echo "kernel=kernel8.img" >> "${BOOT_DIR}"/config.txt
```

Also, I think, that the changes to `config.txt` are not really necessary. `kernel8.img` is the default name for Raspberry Pi 4 (and related models). So, the firmware will automatically boot this kernel.

koehlma, 2024-09-04:
Yes, the reason is that there is no clear place where the boot partition should be mounted as it could be `/boot` or `/boot/firmware`, depending on the OS. I still consider this a bug in the documentation, though, as it is currently undocumented. The core recipes always copy everything to "${BOOT_DIR}" as described above.

closed 2024-09-04 by koehlma

dvorak0, 2024-09-04:
Just letting you know, copying to "${BOOT_DIR}" solved my problem. 

Thanks

commit 4c5e70d7 (2024-09-04): docs: document how to modify the boot partition (fix #31)