rugix/rugix #32: Correct value for "layout" parameter in v0.7?

raryanpur, 2024-09-12:
Hi there,

I'm in the process of migrating from v0.6 to v0.7, and am having trouble specifying the correct value for the layout parameter. For reference, here's `rugpi-bakery.toml`

```
[repositories]
rugpi-extra = { git = "https://github.com/silitics/rugpi-extra.git", branch = "v0.7" }

[images.tryboot]
layer = "customized"
architecture = "arm64"
layout = "pi_image_layout"

# A specific image including the firmware update for Raspberry Pi 4 and CM4.
[images.pi4]
layer = "customized"
architecture = "arm64"
layout = "pi_image_layout"
```

From reading the v0.7 release notes I didn't think I needed to specify either `architecture` or `layout`, but building with the official Rugpi Docker images in CI throws errors without them. In particular when not including `layout`, the following error is thrown:

```
/usr/bin/apt-get
 INFO packing system files
 INFO Extracting layer.
 INFO Creating config and boot directories.
 INFO Initialize boot flow.
Error: image layout needs to be specified
```

Separately, it may be worthwhile updating the `rugpi-template` in the associated repo. At present I don't think that it works with v0.7.

koehlma, 2024-09-12:
> From reading the v0.7 release notes I didn't think I needed to specify either `architecture` or `layout`

The `architecture` must be explicitly specified now. This change somehow slipped through the cracks. I will add this point to the release notes and the migration guide.

Specifying `layout` is not necessary. You have to specify a `target`, though. Likely it is:

```
target = "rpi-tryboot"
```

For details, have a look here: https://oss.silitics.com/rugpi/docs/upgrading-from-v0.6

Does that resolve your issues?

> Separately, it may be worthwhile updating the `rugpi-template` in the associated repo.

I will archive this repository. It is no longer relevant. There are now multiple templates and a new project can be initialized with `./run-bakery init`. For details, see: https://oss.silitics.com/rugpi/docs/getting-started#initializing-the-project

raryanpur, 2024-09-12:
> The `architecture` must be explicitly specified now. This change somehow slipped through the cracks. I will add this point to the release notes and the migration guide.

No worries, thanks for clarifying!

> target = "rpi-tryboot"

Perfect, that did it.

> I will archive this repository. It is no longer relevant. There are now multiple templates and a new project can be initialized with ./run-bakery init. For details, see: https://oss.silitics.com/rugpi/docs/getting-started#initializing-the-project

👍🏼

closed 2024-09-12 by raryanpur