rugix/rugix #34: Compatibility with OrbStack

lukechilds, 2024-10-04:
We recently moved to OrbStack on our development machines but aren't able to run Rugpi builds with it:

```
ghcr.io/silitics/rugpi-bakery@sha256:ed8b522d3511434dc351bc6576bb56d9fbbc4b596f4189bd487ae9db40789901: Pulling from silitics/rugpi-bakery
Digest: sha256:ed8b522d3511434dc351bc6576bb56d9fbbc4b596f4189bd487ae9db40789901
Status: Image is up to date for ghcr.io/silitics/rugpi-bakery@sha256:ed8b522d3511434dc351bc6576bb56d9fbbc4b596f4189bd487ae9db40789901
docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: ioctl(setctty): operation not permitted: unknown.
```

This is potentially more of an issue on OrbStack's side than Rugpi but just making you aware. Not a huge deal since we can switch back to the Docker Desktop context for Rugpi builds but would be great to not have to do that. Also potentially https://github.com/silitics/rugpi/issues/26 or https://github.com/silitics/rugpi/issues/27 serve as workarounds for this.

Keep up the awesome work!

koehlma, 2024-10-04:
One thing you could try is removing the `-it` flag from the Docker flags. The error message seems to suggest that there is a problem with setting the terminal. I'm using Lima-based Docker setup.

lukechilds, 2024-10-04:
Thanks! That gets us a bit further but seems like there's now an issue creating a loop device, potentially https://github.com/silitics/rugpi/issues/26 is a solution.

```
ghcr.io/silitics/rugpi-bakery@sha256:ed8b522d3511434dc351bc6576bb56d9fbbc4b596f4189bd487ae9db40789901: Pulling from silitics/rugpi-bakery
Digest: sha256:ed8b522d3511434dc351bc6576bb56d9fbbc4b596f4189bd487ae9db40789901
Status: Image is up to date for ghcr.io/silitics/rugpi-bakery@sha256:ed8b522d3511434dc351bc6576bb56d9fbbc4b596f4189bd487ae9db40789901
 INFO baking image `pi4`
 INFO baking layer `umbrelos-rugpi`
 INFO baking layer `umbrelos-base`
 INFO Copying root filesystem "/project/build/umbrelos-base.tar"
 INFO extracting system files
 INFO [ 1/7] prepare umbrelOS base image for Rugpi {}
 INFO     - 00-install.sh
 INFO [ 2/7] install and configure Rugpi Ctrl {"rugpi_admin": "false"}
 INFO     - 00-packages
debconf: delaying package configuration, since apt-utils is not installed
E: Unlocking the slave of master fd 16 failed! - unlockpt (22: Invalid argument)
 INFO     - 01-run.sh
 INFO     - 02-run.py
 INFO     - 03-install.sh
        not a dynamic executable
 INFO [ 3/7] setup Rugpi for umbrelOS {}
 INFO     - 00-install.sh
 INFO [ 4/7] install Rugpi update module for Mender {}
 INFO     - 00-packages
 INFO     - 01-install.sh
 INFO [ 5/7] globally patch the `reboot` binary {}
 INFO     - 00-install.sh
 INFO [ 6/7] fix `/etc/hostname` and `/etc/hosts` {}
 INFO     - 00-install.sh
 INFO [ 7/7] cleanup and restore original umbrelOS configuration {}
 INFO     - 00-install.sh
 INFO packing system files
 INFO creating image (size: 4181397504 bytes)
losetup: /dev/loop0: failed to set up loop device: No such file or directory
Error: error running command `losetup -P /dev/loop0 build/umbrelos-pi4.img`: command failed with non-zero exit code 1
=== STDERR ===
losetup: /dev/loop0: failed to set up loop device: No such file or directory
```

lukechilds, 2024-10-04:
> I'm using a Lima-based Docker setup.

Just for some more info, our use case for OrbStack is a Docker implementation for macOS that allows accessing container IPs directly from the macOS host.

koehlma, 2024-10-04:
> Thanks! That gets us a bit further but seems like there's now an issue creating a loop device, potentially #26 is a solution.

Actually, I think, upgrading to version 0.7 should already solve this issue, as it no longer requires a loop device to create the image (it still requires privileges for other operations like chroot, but these seem to work with your setup).

lukechilds, 2024-10-04:
That seems like it does it, although looks like we need some config changes to work with 0.7. I got it half working after some config tweaks getting quite far in the process to `Error: image layout needs to be specified`.

To just quickly confirm I created the a new 0.7 project with:

```bash
./run-bakery init rpi-debian && ./run-bakery bake image tryboot
```

and confirm it completed the build ok so seems like 0.7 does indeed work ok with OrbStack! I'll close this issue, thanks for your help @koehlma!

closed 2024-10-04 by lukechilds

lukechilds, 2024-10-04:
Just FYI when setting `target` the build fails at

```
 INFO Mounting "/tmp/.tmpNu2sk4" to "/tmp/.tmpNu2sk4/roots/system/run/rugpi/bakery/bundle/".
/usr/bin/apt-get
 INFO packing system files
 INFO Extracting layer.
 INFO Creating config and boot directories.
 INFO Initialize boot flow.
 INFO Copy second stage boot scripts.
 INFO Computing partition table.
 INFO Allocating image file.
 INFO Writing image partition table.
label: dos
label-id: 0xdd4c216e
1: start=2048,size=524288,type=0c
2: start=526336,size=262144,type=0c
3: start=788480,size=262144,type=0c
4: start=1050624,size=9025536,type=05
5: start=1052672,size=9020413,type=83

 INFO Patching boot configuration.
Error: No such file or directory (os error 2)
```

So potentially there's still another OrbStack compatibility issue.

koehlma, 2024-10-05:
This is likely caused by a change of the path of the boot partition, see:

https://oss.silitics.com/rugpi/docs/guide/system-customization#boot-partition

You need to modify the recipes accordingly too move all files there.