rugix/rugix #39: Configure a database

davidlekve, 2024-11-27:
I was wandering if there is a good way to configure services as databases as these often must be running on the system to easily configure them using cli or api tools.

What we have tried now includes commands in XX-install scripts as follows:

```
# Create a superuser and a database
su - postgres -c "psql -c \"CREATE ROLE superuser WITH LOGIN PASSWORD 'mypassword' SUPERUSER;\""
su - postgres -c "psql -c \"CREATE DATABASE mydatabase OWNER superuser;\""
```

But these give errors as:
```
psql: error: connection to server on socket "/var/run/postgresql/.s.PGSQL.5432" failed: No such file or directory
	Is the server running locally and accepting connections on that socket?
psql: error: connection to server on socket "/var/run/postgresql/.s.PGSQL.5432" failed: No such file or directory
	Is the server running locally and accepting connections on that socket?
```

We have had similar issues with our influx-database and have ended up with setting up a database and copying the .bolt file which includes the database configuration. This works, but is very opaque.

Solutions we are considering include:
- Making a startup script that does the configuration, but we would prefer to have this done during build.
- Is it so that the image runs in a container and one could connect to it in a XX-run script and configure it from there?

Awesome working with your software!

davidlekve, 2024-12-05:
Sorry for late response, but this totally works. Thank you for response.

If others are having the same issue we ended up using:

`pg_ctlcluster "$VERSION" "$CLUSTER_NAME" start`
and
`influxd --bolt-path=/var/lib/influxdb/influxd.bolt &`

combined with a wait for service to start, excecution of configuration and in the end stopping the "service":

## Influx:
```
#!/bin/bash

# Start InfluxDB as a background process
influxd --bolt-path=/var/lib/influxdb/influxd.bolt &

## Capture the process ID of influxd
INFLUXD_PID=$!

# Wait for InfluxDB to become ready
echo "Waiting for InfluxDB to start..."
while ! curl -s http://localhost:8086/health | grep -q "\"status\":\"pass\""; do
    sleep 1
done
echo "InfluxDB is ready."

# Declare environment variables
INFLUX_URL="http://localhost:8086"
INFLUX_TOKEN=""
INFLUX_ORG=""
INFLUX_BUCKET=""
INFLUX_USERNAME=""
INFLUX_PASSWORD=""

chown influxdb:influxdb /var/lib/influxdb/influxd.bolt
# Run InfluxDB setup
influx setup --username "$INFLUX_USERNAME" \
    --password "$INFLUX_PASSWORD" \
    --org "$INFLUX_ORG" \
    --bucket "$INFLUX_BUCKET" \
    --token "$INFLUX_TOKEN" \
    --retention 0 \
    --force || {
    echo "[ERROR]: InfluxDB setup failed. Ensure InfluxDB is running."
    kill $INFLUXD_PID
    exit 1
}

echo "InfluxDB setup completed successfully."

# Perform any additional tasks here...

# Stop InfluxDB
echo "Stopping InfluxDB..."
kill $INFLUXD_PID

# Wait for the process to exit
wait $INFLUXD_PID
echo "InfluxDB stopped successfully."
```

## Postgres:
```
#!/bin/bash

# Variables
CLUSTER_NAME="main"
VERSION="15" # PostgreSQL version to use
DATA_DIR="/var/lib/postgresql/$VERSION/$CLUSTER_NAME"
LOG_FILE="/var/log/postgresql/postgresql-$VERSION-$CLUSTER_NAME.log"
ROLE_NAME=""
ROLE_PASSWORD=""
DATABASE_NAME=""

# Ensure PostgreSQL is installed
if ! dpkg -l | grep -q postgresql-common; then
    echo "[ERROR]: postgresql-common package is not installed. Please install it using 'apt install postgresql-common'." >&2
    exit 1
fi

# Create a new PostgreSQL cluster if it does not exist
if ! pg_lsclusters | grep -q "$VERSION.*$CLUSTER_NAME"; then
    echo "Creating PostgreSQL cluster '$CLUSTER_NAME' for version '$VERSION'..."
    pg_createcluster "$VERSION" "$CLUSTER_NAME" --datadir="$DATA_DIR"
    if [ $? -ne 0 ]; then
        echo "[ERROR]: Failed to create PostgreSQL cluster '$CLUSTER_NAME'." >&2
        exit 1
    fi
fi

# Start the PostgreSQL cluster
echo "Starting PostgreSQL cluster '$CLUSTER_NAME'..."
pg_ctlcluster "$VERSION" "$CLUSTER_NAME" start
if [ $? -ne 0 ]; then
    echo "[ERROR]: Failed to start PostgreSQL cluster '$CLUSTER_NAME'." >&2
    exit 1
fi

# Wait for PostgreSQL to become ready
echo "Waiting for PostgreSQL to become ready..."
until pg_isready -q -d "postgresql:///$DATABASE_NAME"; do
    sleep 1
done
echo "PostgreSQL is ready."

# Create role
echo "Creating role '$ROLE_NAME'..."
su - postgres -c "psql -c \"CREATE ROLE $ROLE_NAME WITH LOGIN PASSWORD '$ROLE_PASSWORD' SUPERUSER;\""
if [ $? -ne 0 ]; then
    echo "[ERROR]: Failed to create role '$ROLE_NAME'." >&2
    pg_ctlcluster "$VERSION" "$CLUSTER_NAME" stop
    exit 1
fi

# Create database
echo "Creating database '$DATABASE_NAME' owned by '$ROLE_NAME'..."
su - postgres -c "psql -c \"CREATE DATABASE $DATABASE_NAME OWNER $ROLE_NAME;\""
if [ $? -ne 0 ]; then
    echo "[ERROR]: Failed to create database '$DATABASE_NAME'." >&2
    pg_ctlcluster "$VERSION" "$CLUSTER_NAME" stop
    exit 1
fi

# Perform additional tasks
echo "PostgreSQL setup completed successfully."
echo "Role: $ROLE_NAME"
echo "Password: $ROLE_PASSWORD"
echo "Database: $DATABASE_NAME"

# Stop PostgreSQL cluster
echo "Stopping PostgreSQL cluster '$CLUSTER_NAME'..."
pg_ctlcluster "$VERSION" "$CLUSTER_NAME" stop
if [ $? -ne 0 ]; then
    echo "[ERROR]: Failed to stop PostgreSQL cluster '$CLUSTER_NAME'." >&2
    exit 1
fi
echo "PostgreSQL cluster stopped successfully."

exit 0
```

closed 2024-12-05 by davidlekve

koehlma, 2024-12-09:
Great that this worked and thank you for providing the scripts!

Are the scripts contributions, as defined in the Apache 2.0 license, to this project (see project README)? Because, if they are, I would like to add them to the documentation.