rugix/rugix #42: /root/.cache/ not always populated after image build

francinv, 2025-01-22:
I use rugpi for building custom RPi images. One of the recipes is building and installing a python application. To manage dependencies for the application in question I use `poetry`. 

As a part of the recipe I both install `poetry` and also do a `poetry install`. The dependencies are then downloaded to `/root/.cache/pypoetry/xxxx`. My problem is that sometimes this works and all dependencies are available at runtime and other times it does not work. 

I have not unfortunately been able to find some consistency between what I do and when the dependencies are not available at runtime.

Have anyone seen these kind of problems?

Alle help and discussions are greatly appreciated! ☺ 

### Additional Context
- Using Raspberry Pi Imager to upload the image
- Using a Raspberry Pi 5 8GB

koehlma, 2025-01-22:
Do you use the `core/persist-root-home` recipe as part of the build?

Relying on a cache may also not be the best idea in general. Can't you install the application properly, e.g., by creating a Python virtual environment for your application under `/opt` that then contains all the dependencies?

francinv, 2025-01-22:
Thank you for a quick response 🙌🏾 

> Do you use the `core/persist-root-home` recipe as part of the build?
Yes, I do.

> Relying on a cache may also not be the best idea in general. Can't you install the application properly, e.g., by creating a Python virtual environment for your application under `/opt` that then contains all the dependencies?
Can of course do that. I have favorized `poetry` for local dev, but a lot of headache when building an image.

I think I will test a Python virtual environment. Or maybe move where Poetry creates the environment as this is stored under `.cache`.

koehlma, 2025-01-22:
> Thank you for a quick response 🙌🏾

You are welcome! Out of curiosity, is this a hobby project or are you building some sort of product?

> Yes, I do.

Could it be that this works when you have a fresh deployment but stops working when doing an update? This recipe will install a configuration file that persists anything under `/root` on the data partition. As a result, you will get the previous version of the cache when doing an update.

Yeah, I think Poetry for local development is great, but for production, you would want a proper installation in a dedicated virtual environment.

francinv, 2025-01-22:
> You are welcome! Out of curiosity, is this a hobby project or are you building some sort of product?

Building a product! I am pretty sure you have had a meeting with my colleague David. We are from Norway and building our startup enmo :)

> Could it be that this works when you have a fresh deployment but stops working when doing an update? This recipe will install a configuration file that persists anything under `/root` on the data partition. As a result, you will get the previous version of the cache when doing an update.

Could be! Testing this now.

francinv, 2025-01-23:
Closing this issue as using Python virtual environment solved all my problems. Even issues related to other stuff as well 😂

closed 2025-01-23 by francinv