rugix/rugix #47: rugix-bakery build cache is disabled or broken

reubenmiller, 2025-02-13:
In rugix v0.8, the caching mechanism seems to be disabled or broken. When building the same twice results in downloading the same base image.

For instance, using [rugix-bakery.toml](https://github.com/thin-edge/tedge-rugpi-core/blob/v0.8-rugix/rugix-bakery.toml) in the thin-edge.io project, running the following commands invokes the error scenario:

```sh
./run-bakery bake image raspios-tedge-pi5

# The second build downloads the same base .img.xz again
./run-bakery bake image raspios-tedge-pi5
```

After the first build, the `.rugix/cache` folder does contain the expanded file but it just does not seem to be used by rugix-bakery.

Below shows the `.rugix/cache` after the first build.

```sh
$ ls -l .rugix/cache 
total 5727552
-rw-r--r--  1 reubenmiller  staff  2835349504 Feb 13 20:51 76c24acfa10766a9afb9cbf1ebacd59417010fc2.img
-rw-r--r--  1 reubenmiller  staff   452730060 Feb 13 20:49 76c24acfa10766a9afb9cbf1ebacd59417010fc2.img.xz
```

To rule our a problem with the `.rugix` folder permissions, I deleted the folder and repeated the test, however the same error behaviour was observed.

**Version info**

|Rugix|-|
|--|--|
|Image |ghcr.io/silitics/rugix-bakery:v0.8|
|Digest|sha256:120a379fd997e4bf65074cebfe4317ce857f322b431e7385a10b8662c52fabc5|

### Detailed console output

Building the image the first time will download the image (`*.img.xz`).

```sh
$ ./run-bakery bake image raspios-tedge-pi5

bash: warning: setlocale: LC_ALL: cannot change locale (C.UTF-8): No such file or directory
v0.8: Pulling from silitics/rugix-bakery
Digest: sha256:120a379fd997e4bf65074cebfe4317ce857f322b431e7385a10b8662c52fabc5
Status: Image is up to date for ghcr.io/silitics/rugix-bakery:v0.8
ghcr.io/silitics/rugix-bakery:v0.8
 INFO baking image `raspios-tedge-pi5`
 INFO baking layer `raspios-tedge`
 INFO baking layer `raspios`
 INFO downloading `https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2024-07-04/2024-07-04-raspios-bookworm-arm64-lite.img.xz`
 INFO decompressing XZ image

 ...

 INFO Patching boot configuration.
 INFO Patching `config.txt`.
 INFO Creating fat32 filesystem on partition 1 (size: 524288).
 INFO Creating fat32 filesystem on partition 2 (size: 262144).
 INFO Creating ext4 filesystem on partition 5 (size: 3820870).
```

Then running the same task again, the `*.img.xz` file is downloaded again.

```sh
$ ./run-bakery bake image raspios-tedge-pi5

bash: warning: setlocale: LC_ALL: cannot change locale (C.UTF-8): No such file or directory
v0.8: Pulling from silitics/rugix-bakery
Digest: sha256:120a379fd997e4bf65074cebfe4317ce857f322b431e7385a10b8662c52fabc5
Status: Image is up to date for ghcr.io/silitics/rugix-bakery:v0.8
ghcr.io/silitics/rugix-bakery:v0.8
 INFO baking image `raspios-tedge-pi5`
 INFO baking layer `raspios-tedge`
 INFO baking layer `raspios`
 INFO downloading `https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2024-07-04/2024-07-04-raspios-bookworm-arm64-lite.img.xz`
```

reubenmiller, 2025-02-13:
It seems to be only visible when using images like raspio, e.g.

```sh
name = "Raspberry Pi OS (Bookworm)"
url = "https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2024-07-04/2024-07-04-raspios-bookworm-arm64-lite.img.xz"
```

koehlma, 2025-02-14:
Thanks for reporting! The cache was always cleared regardless whether the Docker image has changed. Should be fixed in version 0.8.1.

commit 4f839699 (2025-02-14): fix: only clear cache when Docker image is newer (fixes #47)
closed 2025-02-14 by koehlma