rugix/rugix #48: Build error: Could not allocate block

francinv, 2025-02-16:
I am trying to build a rugpi image and are constantly getting this error message after some change to one recipe:

```bash
__populate_fs: Could not allocate block in ext2 filesystem while writing file "postgresql-client-common.list"
mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system
Error: error running command `mkfs.ext4 -d /tmp/.tmpcjOM1r/roots/system /tmp/.tmpcjOM1r/ext4.img`: command failed with non-zero exit code 1
=== STDERR ===
mke2fs 1.47.0 (5-Feb-2023)
__populate_fs: Could not allocate block in ext2 filesystem while writing file "postgresql-client-common.list"
mkfs.ext4: Could not allocate block in ext
```

If I revert the changes the build works fine. But I do not understand what changes I have done can lead to this?

The recipe in question downloads a React/Next.JS application to the image. Was previously a Svelte application so not many changes expect the actual files for the project.

Anybody have an idea of what the reason can be? 🤔

francinv, 2025-02-16:
Think the issue is related to either file size or amount of files. By upgrading to `v0.8` I was able to build successfully. 

However I am now not able to boot my Raspberry Pi 5 with the new image:
[image].

Can this be related to either size or amount of files?

koehlma, 2025-02-16:
Yes, that is related to the number of files/system partition size. Version 0.8 includes a higher overhead factor to calculate the filesystem size, that's why it is working there (we need to make this configurable eventually). The error when trying to boot is a bit concerning (it tries to write a partition table that seems incompatible with the existing layout), I have to look into that. By default, the size of the system partition is just 4GB. If you need more than that, you need to configure a different size in `bootstrapping.toml` as described here: https://oss.silitics.com/rugix/docs/ctrl/bootstrapping#default-layout

davidlekve, 2025-02-16:
Got through the boot with larger system-size :)

koehlma, 2025-02-16:
> Could problems arise when doing OTA from a system with 4GB partitions to one that needs e.g 6GB or will rugix handle this?

Such an update will fail. Handling this would require changing the partition table and resizing filesystem which is rather risky in production. So, while it is in principle doable, Rugix does not do it.

davidlekve, 2025-02-16:
Can we solve this without changing the system-size and then doing normal ota to production? Do you have recommendations?

koehlma, 2025-02-17:
Just to clarify, for the purposes of the build and update installation, the configured `system-size` is actually irrelevant, only the actual size of the root filesystem matters. The `system-size` configuration is only used during the bootstrapping phase and otherwise has no effect. That means, that you can build an update with a changed `system-size` and _try_ to install it on any device with a larger or smaller system size and this installation may succeed if the root filesystem fits onto the existing system partitions. However, doing this will not change the size of the system partitions as initially provisioned.

The problem is that if your updated root filesystem actually takes more space than is available on a given system partition, then there is no way to install this update without growing the system partitions on the device. The only other option is to make the update fit by reducing its size (or using a filesystem with compression, but this is currently not supported by Rugix and may also have other downsides). If reducing the size of the update is not an option, there is, unfortunately, no way around growing the system partitions. There are different ways to grow the system partitions, depending on how much you care about the data stored on the device. **If you don't care about the data at all**, the easiest way would be as follows:

1. Build an update that does fit and specifies the new large `system-size`.
2. Make sure that this update is installed to the A partition, e.g., by installing it twice.
3. Place the bootstrap marker `/.rugix/bootstrap` on the config partition mounted at `/run/rugix/mounts/config`. Note that this requires remounting this partition as read-write as it is usually write protected.
4. Delete the data and second system partition without reloading the partition table (reloading should be impossible anyway as the partitions are in-use).
5. Reboot.

When you reboot, Rugix will expand the first system partition and then recreate the second system partition and the data partition (it will bootstrap the device again). There might be a slim **chance for issues if this process is interrupted**, e.g., due to a power outage between step 3 and 4, as well as when the bootstrapping process is interrupted. Growing the partitions that way constitutes a complete reset of the device and **all data will be lost**. This includes SSH host keys and the `machine-id` of the system (which are stored on the data partition). Also, **I did not test this process**. If you only care about a small amount of crucial data, you could place it on the config partition or on the boot partitions and later restore it.

The other option is to shrink the data partition to make room for the larger system partitions. However, this cannot be done in a safe way. When shrinking the filesystem on the data partition, **any interruption will likely result in a complete loss of all data and will render the device inoperable/unbootable**. Also, this can only be done while the data partition is not in use which means that the overlay must be temporarily disabled or placed in memory and all data directories persisted with Rugix's state management mechanism must be unmounted. There may be ways to make this more safe, again by using a more complex process and storing crucial data on other partitions and generally instrumenting the boot process to be able to deal with a broken data partition. Generally, crucial data should be stored redundantly and there should be backups. So, one could also consider a process where a user first creates a backup of some kind to an external storage medium before attempting such an update. I cannot give you any more specific recommendation for how to proceed without considering the particularities of your setup and the requirements you have with respect to device data. There are many options with different tradeoffs. I think, if one has a significant number of devices in production and runs into this problem, then this will always require some dedicated engineering (which I would be happy to provide as part of a paid consulting engagement).

I hope that this helps.

koehlma, 2025-02-17:
I did add a [Production Checklist](https://oss.silitics.com/rugix/docs/production-checklist) to the docs, explicitly calling out that one should ensure that the system size will be sufficient when the system evolves. Really don't want people to run in this issues with insufficiently-sized system partitions because it is unfortunately very difficult to resolve.

closed 2025-02-26 by francinv