rugix/rugix #49: starting docker service in vm interrupts ssh connection making vm unreachable

reubenmiller, 2025-02-18:
Note: This may not be directly related to rugix, but just sharing it in case if it could be of interest

### Description

If an image is built which containers docker (~v27.x) and the image is used to spin up a virtual machine via the `./run-bakery run` command, then the ssh console stops functioning after the `docker.service` starts.

I suspect that starting the docker some network filtering (via iptables) is being added which is then killing the ssh connection (and preventing new ssh connections).

On the vm's output, there is a warning about not having the `br_netfilter` kernel module loaded by default, however I tried a fe

```sh
> Debian GNU/Linux 12 unknown-525400123456 ttyAMA0                          
>
> unknown-525400123456 login: [  330.251088] audit: type=1400 audit(1739914899.500:9): apparmor="STATUS" operation="profile_load" profile...
> [  335.795978] Initializing XFRM netlink socket                                                                                        
> [  335.965951] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if yo...
```

### Version info

|Rugix|-|
|--|--|
|Image |ghcr.io/silitics/rugix-bakery:v0.8|
|Digest|sha256:120a379fd997e4bf65074cebfe4317ce857f322b431e7385a10b8662c52fabc5|
|Host|MacOS (m1)|
|Target Build System|RaspOS (Debian bookworm)|
|Target Arch|arm64 (though other architectures are probably also affected)|

### Steps to reproduce

It is assumed that you've already configured your ssh key either in the `rugix-bakery.toml` or in the `.env` file under the variable `SSH_KEYS_<name>=xxxxx`

1. Clone the repo

    ```sh
    git clone https://github.com/thin-edge/tedge-rugpi-core
    ```

    Note: The repository is actually meant to provide recipes to other projects, however it also includes a `rugix-bakery.toml` for testing purposes.

2. Build and run the vm (assuming arm64). This will pull in the `docker` recipe (included in the repo)

    ```sh
    ./run-bakery run debian-bookworm-tedge-efi-arm64
    ```

    Note: The `docker` service is not configured to auto start due to this problem.

3. Connect to the vm via ssh in another console

```sh
ssh -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -p 2222 root@127.0.0.1
```

4. Start the docker service

    ```sh
    systemctl start docker.service
    ```

    Wait about 5-10 seconds, and then notice that the ssh console becomes unresponsive (most likely due to the network connectivity being interrupted)

    Below shows an example of the ssh connection to the vm, and starting docker.

    ```sh
    $ ssh -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -p 2222 root@127.0.0.1
    Warning: Permanently added '[127.0.0.1]:2222' (ED25519) to the list of known hosts.
    Linux unknown-525400123456 6.1.0-31-arm64 #1 SMP Debian 6.1.128-1 (2025-02-07) aarch64

    The programs included with the Debian GNU/Linux system are free software;
    the exact distribution terms for each program are described in the
    individual files in /usr/share/doc/*/copyright.

    Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
    permitted by applicable law.
    root@unknown-525400123456 ~ # systemctl start docker.service

    # Connection is dropped after ~10 seconds
    ```

### References

Some of the links found during some research which might be useful (though I didn't have luck with any of the suggested "fixes"):

* https://github.com/moby/moby/discussions/48559
* https://bbs.archlinux.org/viewtopic.php?id=280798
* https://github.com/blueman-project/blueman/issues/739

reubenmiller, 2025-02-19:
Note: If I use the `podman` recipe (also included in the same tedge-rugpi-core repo), then the VM does work fine (even after running a container). I know docker and podman do things very differently, but at least it confirms that conterization within the VM is not incompatible.

koehlma, 2025-02-23:
I think that the problem is not directly related to the Kernel message:

```
bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
```

You can provoke this message with `modprobe -va bridge br_netfilter` but it will not trigger the issue. Also, if I boot an image with UTM (using basically the same settings as Rugix Bakery to set up networking), it shows the same messages but the SSH access via the port forwarding (from QEMU to host) continues to work fine. I will investigate this further.

koehlma, 2025-02-27:
There seems to be an issue with QEMU user mode networking and routing. My guess is that this is somehow caused by the fact that the outside Docker network and the inside Docker network share the same IP address ranges.

As a temporary fix, you can overwrite the networking mode on the host:

```
DOCKER_FLAGS="--network host" ./run-bakery ...
```

Note that this renders the port forwarding through `-p` ineffective and all ports are instead directly exposed on the host. If you are running Docker through a VM (e.g., on macOS), the port will no longer be wired through to your host. You can, however, spawn a shell with the `shell` subcommand and then connect via SSH from there:

```
DOCKER_FLAGS="--network host" ./run-bakery shell
```

koehlma, 2025-02-27:
As another alternative, one can also create a Docker network without IPv4 enabled:

```
docker network create --ipv6 -o com.docker.network.enable_ipv4=false rugix-net
```

This network can then be used with the following `DOCKER_FLAGS`:

```
DOCKER_FLAGS="--network=rugix-net"
```

It might make sense to add something like that tho `run-bakery`.

reubenmiller, 2025-03-01:
Finally found some time to try it out and both solutions worked fine (I'll probably adopt the dedicated docker network option by default though). Thanks for spending the time to look into this.

I'm ok with closing the ticket, or do you want to leave it open for some built-in feature (that would be a nice-to-have, not a must in my view)?

koehlma, 2025-03-01:
Let's leave it open. If there are no downsides to the network approach, I am inclined to include it in the `run-bakery` script. Otherwise users may run into that problem and have to search for the solution. I would like to avoid that.

reubenmiller, 2025-03-20:
> Did you notice any downsides of the network solution so far? If not, I think, it would make sense to make that the default.

Generally I think it works fine...the only weirdness I've seen is that if I start the vm, forget about it for ~30 mins, then try to connect via ssh, then the connection is refused. I haven't spent the time yet to create a meaningful ticket (e.g. after how long, does it occur each time etc.).