rugix/rugix #51: rugix-ctrl system info only ouputs json format if stderr is redirected

lyracrandall, 2025-02-25:
Hello!

While trying to write an update script, I decided to start using the install-update.sh script given in the documentation as a template (https://oss.silitics.com/rugix/docs/ctrl/over-the-air-updates). However, I noticed that "rugix-ctrl system info" only ever gave this output, instead of the seemingly expected json output and I couldn't pipe any information out of it either:

```
root@test-os:~ $ rugix-ctrl system info
Boot Flow: tryboot
Active Boot Group: a
Default Boot Group: a
Slot "boot-a": active
Slot "boot-b": inactive
Slot "system-a": active
Slot "system-b": inactive
```

However, it does give json output if I redirect stderr to /dev/null:

```
root@test-os:~ $ rugix-ctrl system info 2> /dev/null
{ "slots":{"boot-a":{"active":true},"boot-b":{"active":"false"},"system-a":{"active":true},"system-b":{"active":false}}, "boot":{"bootFlow":"tryboot","activeGroup":"a", "defaultGroup":"a", "groups":{"a":{},"b":{}}}}
```

Is this a bug? Or is this intended behavior?

koehlma, 2025-02-25:
Thanks for reporting!

Indeed, `rugix-ctrl` wrongly checked whether stderr is piped instead of checking stdout.

lyracrandall, 2025-02-25:
Thank you so much!

commit 579897af (2025-02-25): fix: check whether `stdout` is piped instead of `stderr` (fixes #51)