rugix/rugix #62: Bake Error: volume with name rugix-build-cache already exists: volume already exists

ben-auo, 2025-03-25:
Very cool project! I tried following your Quickstart and ran into an error when attempting to bake:

```
# whoami
root
# cat /etc/os-release | grep UBUNTU_VERSION=
UBUNTU_VERSION="24.04.2 LTS (Noble Numbat)"
# podman version
Client:       Podman Engine
Version:      4.9.3
API Version:  4.9.3
Go Version:   go1.22.2
Built:        Wed Dec 31 17:00:00 1969
OS/Arch:      linux/amd64
# mkdir example && cd example && pwd
/root/example
# curl -sfSO https://raw.githubusercontent.com/silitics/rugix/v0.8/bakery/run-bakery && chmod +x ./run-bakery
# ./run-bakery init quickstart-guide
Trying to pull ghcr.io/silitics/rugix-bakery:v0.8...
Getting image source signatures
Copying blob 59554daaff16 done   | 
Copying blob 155ad54a8b28 done   | 
Copying blob b53c0143e9cb done   | 
Copying blob 1f01cd3ccae7 done   | 
Copying blob 9cd72aa261c3 done   | 
Copying blob 1a4aad3c4785 done   | 
Copying blob 415a9300317f done   | 
Copying blob 020b72b1beaf done   | 
Copying blob fbd318c08281 done   | 
Copying blob a29d4b74f498 done   | 
Copying blob b4d727a5978f done   | 
Copying blob 531069f457f3 done   | 
Copying blob 1b59c9a2f04b done   | 
Copying blob bf322b903fd4 done   | 
Copying blob aafecfe310f1 done   | 
Copying config 7ee944013b done   | 
Writing manifest to image destination
7ee944013b40bd76dd3668533a4799ada999354cdb8bb2aae151b40e0f920737
# sed -i 's/Hello/Hi/g' recipes/hello-world/html/index.html
# nano layers/customized.toml
# ...added my public key
# ./run-bakery bake image customized-efi-amd64
Trying to pull ghcr.io/silitics/rugix-bakery:v0.8...
Getting image source signatures
Copying blob 59554daaff16 skipped: already exists  
Copying blob 155ad54a8b28 skipped: already exists  
Copying blob 020b72b1beaf skipped: already exists  
Copying blob b53c0143e9cb skipped: already exists  
Copying blob 9cd72aa261c3 skipped: already exists  
Copying blob 1a4aad3c4785 skipped: already exists  
Copying blob 415a9300317f skipped: already exists  
Copying blob 1f01cd3ccae7 skipped: already exists  
Copying blob b4d727a5978f skipped: already exists  
Copying blob 531069f457f3 skipped: already exists  
Copying blob fbd318c08281 skipped: already exists  
Copying blob 1b59c9a2f04b skipped: already exists  
Copying blob bf322b903fd4 skipped: already exists  
Copying blob a29d4b74f498 skipped: already exists  
Copying blob aafecfe310f1 skipped: already exists  
Copying config 7ee944013b done   | 
Writing manifest to image destination
7ee944013b40bd76dd3668533a4799ada999354cdb8bb2aae151b40e0f920737
Error: volume with name rugix-build-cache already exists: volume already exists
```

koehlma, 2025-03-27:
There seems to be some inconsistency between Docker and Podman. I slightly modified the `run-bakery` script to account for that. Can you please update the script with the following command and then try again:

```shell
curl -sfSO https://raw.githubusercontent.com/silitics/rugix/main/bakery/run-bakery && chmod +x ./run-bakery
```

Note that this is different from the command from the docs in that it uses the main branch, not the v0.8-tagged script.

ben-auo, 2025-03-28:
New error with that version:

```
$ ./run-bakery bake image customized-efi-amd64
Trying to pull ghcr.io/silitics/rugix-bakery:v0.8...
Pulling image //ghcr.io/silitics/rugix-bakery:v0.8 inside systemd: setting pull timeout to 5m0s
Getting image source signatures
Copying blob 390fee8a2546 skipped: already exists  
Copying blob c8935da17dbd skipped: already exists  
Copying blob 79808928ce62 skipped: already exists  
Copying blob b73fe1d750b5 skipped: already exists  
Copying blob a901c77e52a4 skipped: already exists  
Copying blob 460ddb759d55 skipped: already exists  
Copying blob 3ff389bdb268 skipped: already exists  
Copying blob 10ae0f4c4aea skipped: already exists  
Copying blob 2086591076ff skipped: already exists  
Copying blob 8a4383fb81c8 skipped: already exists  
Copying blob e6f717792ff0 skipped: already exists  
Copying blob 7cd785773db4 skipped: already exists  
Copying blob 4a7ee39dd2fc skipped: already exists  
Copying blob 9a720272370e skipped: already exists  
Copying blob b1c55ea69442 skipped: already exists  
Copying config 008bc016ad done   | 
Writing manifest to image destination
008bc016ad83f01013fc09fce486673ec50aef73fcef9ce6fa29bbb5355aa79c
 INFO baking image `customized-efi-amd64`
 INFO baking layer `customized-efi`
 INFO baking layer `customized`
 INFO baking layer `debian-bookworm`
 INFO Creating empty layer.
 INFO [ 1/1] bootstrap Debian from scratch {"snapshot": "", "suite": "bookworm"}
 INFO     - 00-run.sh
 INFO packing system files
 INFO Extracting layer.
 INFO [ 1/9] update package lists {}
 INFO     - 00-install.sh
Log: ".rugix/layers/407954042d5dc8d6c7a9d76ca7a64ba3a572f311/build.log"
Error: unable to mount /dev
├╴at crates/tools/rugix-bakery/src/oven/customize.rs:315:62
│   
╰─▶ unable to bind mount
    ├╴at /project/crates/libs/rugix-common/src/mount.rs:68:10
    ├╴dst: "/tmp/.tmpxxg8RP/roots/system/dev"
    ├╴src: "/dev"
    │   
    ╰─▶ EINVAL: Invalid argument
```

koehlma, 2025-03-28:
Can you run `./run-bakery shell` and then run `mount` and post the output? Before the change it worked the first time when running it (after `podman system reset`)? Does it now always fail or still work the first time? Are you running a rootless Podman setup?

ben-auo, 2025-03-30:
Here you go:

```
$ mkdir /tmp/test && cd /tmp/test
$ podman system reset -f
$ curl -sfSO https://raw.githubusercontent.com/silitics/rugix/main/bakery/run-bakery && chmod +x ./run-bakery
$ ./run-bakery shell
Trying to pull ghcr.io/silitics/rugix-bakery:v0.8...
Pulling image //ghcr.io/silitics/rugix-bakery:v0.8 inside systemd: setting pull timeout to 5m0s
Getting image source signatures
Copying blob 390fee8a2546 done   | 
Copying blob 7cd785773db4 done   | 
Copying blob c8935da17dbd done   | 
Copying blob 79808928ce62 done   | 
Copying blob 3ff389bdb268 done   | 
Copying blob a901c77e52a4 done   | 
Copying blob 10ae0f4c4aea done   | 
Copying blob b73fe1d750b5 done   | 
Copying blob 460ddb759d55 done   | 
Copying blob b1c55ea69442 done   | 
Copying blob 2086591076ff done   | 
Copying blob 8a4383fb81c8 done   | 
Copying blob e6f717792ff0 done   | 
Copying blob 4a7ee39dd2fc done   | 
Copying blob 9a720272370e done   | 
Copying config 008bc016ad done   | 
Writing manifest to image destination
008bc016ad83f01013fc09fce486673ec50aef73fcef9ce6fa29bbb5355aa79c
root@297c9cbe8145 /project # mount
udev on /dev type devtmpfs (rw,nosuid,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev,inode64)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,nosuid,nodev,relatime,pagesize=2M)
/dev/mapper/data-root on /project type ext4 (rw,noatime,errors=remount-ro)
udev on /dev/vcsa1 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/i2c-0 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
/dev/mapper/data-root on /run/notify type ext4 (rw,noatime,errors=remount-ro)
tmpfs on /etc/hostname type tmpfs (rw,nosuid,nodev,relatime,size=1626272k,nr_inodes=406568,mode=700,uid=1000,gid=1000,inode64)
udev on /dev/btrfs-control type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
tmpfs on /etc/hosts type tmpfs (rw,nosuid,nodev,relatime,size=1626272k,nr_inodes=406568,mode=700,uid=1000,gid=1000,inode64)
shm on /dev/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=64000k,uid=1000,gid=1000,inode64)
tmpfs on /run/.containerenv type tmpfs (rw,nosuid,nodev,relatime,size=1626272k,nr_inodes=406568,mode=700,uid=1000,gid=1000,inode64)
udev on /dev/zfs type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/cpu_dma_latency type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/cuse type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/dm-0 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/dm-1 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/null type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/zero type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/vhost-vsock type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/vhost-net type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/drm_dp_aux0 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/ecryptfs type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/fb0 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/full type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/fuse type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/hidraw0 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/hpet type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/hwrng type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/nvme0 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/i2c-1 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/i2c-2 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/i2c-3 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/i2c-4 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/vhci type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/vga_arbiter type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/vcsu7 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/nvme0n1 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
udev on /dev/vcsu5 type devtmpfs (rw,nosuid,noexec,relatime,size=8091004k,nr_inodes=2022751,mode=755,inode64)
[... 127 lines cut ...]
[... cut ...] share/containers/storage/overlay/5373f4e3f2c93bba5639c7960289d15e40fcc89bac9db717c2266c9aa82f4022/work,redirect_dir=nofollow,uuid=on,volatile,userxattr)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
devpts on /dev/console type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
root@297c9cbe8145 /project # 

```

ben-auo, 2025-03-30:
> Before the change it worked the first time when running it (after podman system reset)? 

After running podman system reset and using the 0.8 version, the `bake image` command will always work and the `bake bundle` next command will fail.

> Does it now always fail or still work the first time? 

Always fails with the main trunk version (same error as above)

> Are you running a rootless Podman setup? 

No, not running rootless:

```
$ podman run --rm alpine id -u
Resolved "alpine" as an alias (/etc/containers/registries.conf.d/shortnames.conf)
Trying to pull docker.io/library/alpine:latest...
Pulling image //alpine:latest inside systemd: setting pull timeout to 5m0s
Getting image source signatures
Copying blob f18232174bc9 done   | 
Copying config aded1e1a5b done   | 
Writing manifest to image destination
0
```

> For the time being, may I suggest to switch to Docker while we investigate the issue. This might be related to Podman.

Okay

koehlma, 2025-03-30:
> After running podman system reset and using the 0.8 version, the `bake image` command will always work and the `bake bundle` next command will fail.

The error message you shared above seems to be from the `bake image` command. The `bake bundle` command may run `bake image` under the hood but the error message you shared above clearly happens during the `bake image` part, not the `bake bundle` part, so I am a bit confused right now.

ben-auo, 2025-03-30:
Sorry for the confusion. Perhaps this will be clearer:

`silitics/rugix/v0.8/bakery/run-bakery` : can run bake image after podman system reset but then not bake bundle due to 'volume with name rugix-build-cache already exists' error

`silitics/rugix/main/bakery/run-bakery` : cannot run bake image due to 'unable to mount /dev' error

ben-auo, 2025-03-30:
Docker appears to work fine with `silitics/rugix/v0.8/bakery/run-bakery`

```
for pkg in docker.io docker-doc docker-compose docker-ce podman-docker podman-desktop containerd runc; do sudo apt-get remove $pkg; done

sudo apt-get autoremove

sudo rm /etc/apt/sources.list.d/docker.list
sudo rm /etc/apt/keyrings/docker.asc
sudo rm -rf /var/lib/docker
sudo rm -rf /var/lib/containerd
sudo rm -rf /etc/docker
sudo rm -rf /etc/containers
sudo rm -rf /var/run/podman
sudo rm -rf /var/run/containers-daemon

# Add Docker's official GPG key:
sudo apt-get update
sudo apt-get install ca-certificates curl
sudo install -m 0755 -d /etc/apt/keyrings
sudo curl -fsSL https://download.docker.com/linux/ubuntu/gpg -o /etc/apt/keyrings/docker.asc
sudo chmod a+r /etc/apt/keyrings/docker.asc

# Add the repository to Apt sources:
echo \
  "deb [arch=$(dpkg --print-architecture) signed-by=/etc/apt/keyrings/docker.asc] https://download.docker.com/linux/ubuntu \
  $(. /etc/os-release && echo "${UBUNTU_CODENAME:-$VERSION_CODENAME}") stable" | \
  sudo tee /etc/apt/sources.list.d/docker.list > /dev/null
sudo apt-get update

sudo apt-get install docker-ce docker-ce-cli containerd.io docker-buildx-plugin docker-compose-plugin

sudo docker run hello-world

$ Hello from Docker!
$ This message shows that your installation appears to be working correctly.
```

```
mkdir /tmp/test2 && cd /tmp/test2
curl -sfSO https://raw.githubusercontent.com/silitics/rugix/v0.8/bakery/run-bakery && chmod +x ./run-bakery
sudo ./run-bakery init quickstart-guide
sudo sed -i 's/Hello/Hi/g' recipes/hello-world/html/index.html
sudo nano layers/customized.toml
# ...added my public key
sudo ./run-bakery bake image customized-efi-amd64
ls -lh build/customized-efi-amd64/system.img 
-rw-r--r-- 1 root root 2.2G Mar 30 09:13 build/customized-efi-amd64/system.img

sudo sed -i 's/Hi/Hi Again/g' recipes/hello-world/html/index.html
sudo ./run-bakery bake bundle customized-efi-amd64
ls -lh build/customized-efi-amd64/system.rugixb 
-rw-r--r-- 1 root root 379M Mar 30 09:23 build/customized-efi-amd64/system.rugixb
```