rugix/rugix #64: Can't boot raspios-bookworm image on Pi-5

miromad, 2025-04-06:
Hello,
From what I've seen so far Rugix seems to be very cool and promising project.
However I had mixed results and was able to get debian-bookworm working on my HW but I'm still unable to build most simplistic working raspios-bookworm image that boots.

Obviously touch display 2 won't is supported only with latest RaspiOS so I'm currently just trying to get prompt on the console (HDMI monitor) with the raspios-bookworm distro supported out of the box (lite 2014-03-15).

Eventually I'd like to get latest standard RaspIOS distro working (desktop, 2024-11-19).

Below are illustrative configs that I tried (1-5). 
First four resulted in bootable image but fifth didn't.
I didn't get any build errors that would point to the possible issues.
Did I miss some step or is it something else?
Thanks

HW:
- Pi-5 Single Board 64M (Oct/23)
- Touch Display 2 (Nov 4-th/24)
- HDMI Monitor, Keyboard & Mouse
- Ubuntu 24 on amd64 laptop for building images

Raspberry Pi OS standard images:
1. - RaspiOS 2024-11-19 (Desktop)
2. - RaspiOS 2024-11-19 Lite
3. - RaspiOS 2024-03-15 Lite

Rugix 0.8.5 (qickstart-guide setup)
4. - customized-pi5 (core/debian-bookworm)
5. - raspios-pi5 (core/raspios-bookworm)

$ uname -a
Linux P14s 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

layers/raspios-pi.toml:

parent = "core/raspios-bookworm"

recipes = [
    # Prepares RaspiOS for booting on a Raspberry Pi.
    "core/rpi-raspios-setup",
]

rugix-bakery.toml:

[systems.raspios-pi5]
layer = "raspios-pi"
architecture = "arm64"
target = "rpi-tryboot"

reubenmiller, 2025-04-07:
I can confirm that I my Raspberry PI 5 device also has issues when using rugix v0.8.5. It seems like the device isn't able to find either the boot loader or the root filesystem, as when the device boots, the fan goes on 100% for about 30 seconds, then the device reboots, and the cycle repeats.

However I tried building the image with rugix v0.8.4, and the custom image booted ok, so it seems that are some incompatibilities introduced some time after v0.8.4.

You can force a specific version by setting the following env variable in your shell, or adding the variable to the `./run-bakery` file.

```
RUGIX_VERSION="v0.8.4"
```

reubenmiller, 2025-04-07:
After some more testing I've been able to locate the exact version where it start to fail. Below are my test results:

Device: Raspberry Pi 5B (4GB RAM)

|RUGIX_VERSION|Boot Test|
|-------|------|
|v0.8.4|PASS|
|git-5fcfda7|PASS|
|git-63ebb64|FAIL|
|v0.8.5|FAIL|

miromad, 2025-04-08:
I switched to v0.8.4 and made further progress.
Now I see console messages and got as far as:

```
[OK] Started rpc-statd-notify.service - Notify NFS peers of restart.
[OK] Finished systemd-update-utmp-runlevel.service - Record Runlevel change in UTMP.
```

but still no login prompt that I get on "vanilla" `RaspiOS 2024-03-15 Lite` that `core/raspios-bookworm` layer is using.

@reubenmiller I wonder what else is different other than slightly different Pi-5 HW.
Are you using same layers/recipes as in my config 5 ?

koehlma, 2025-04-08:
Thanks for the investigation @reubenmiller. The issue is indeed caused by the following commit: https://github.com/silitics/rugix/commit/266719ca3085c84c92cae82cc8b7398145418eda Previously, we did remove the `auto_initframfs` option from the `config.txt` as it was incompatible with Rugix's pre-init process. We do not do this anymore as it is a prerequisite for using SquashFS on a Raspberry Pi with the Debian-based builds. The issue seems to be that Raspberry Pi's initramfs is incompatible with the boot options we specify. To fix the issue, remove the line starting with `auto_initramfs` from the `config.txt`, e.g., with:

```
sed -i '/^auto_initramfs=/d' "${RUGIX_LAYER_DIR}/roots/boot/config.txt"
```

I will update the `rpi-raspios-cleanup` recipe accordingly.

koehlma, 2025-04-08:
> Obviously touch display 2 won't is supported only with latest RaspiOS so I'm currently just trying to get prompt on the console (HDMI monitor) with the raspios-bookworm distro supported out of the box (lite 2014-03-15).

@miromad: We generally recommend building upon Debian directly. The Debian build includes the official Raspberry Pi repositories. So, you should be able to install whatever is necessary to get everything working. The Raspberry Pi OS distribution is geared towards direct installation and usage and not really suitable for building embedded devices upon. With a Debian-based build, you need to install everything you need explicitly but at the same time get an image that is specially designed for your use case.

commit 3e3803d6 (2025-04-08): fix: remove `auto_initramfs` option (fixes #64)

The initramfs shipped with Raspberry Pi OS is incompatible with the `root=PARTUUID` option we use to specify the system partition.
closed 2025-04-08 by koehlma