rugix/rugix #65: 2.3 GB Alpine Linux?

ben-auo, 2025-04-13:
I didn't expect the Alpine Linux image to be so large. Am I missing anything or is this expected behavior?

```
$ mkdir ~/example && cd ~/example
$ curl -sfSO https://raw.githubusercontent.com/silitics/rugix/v0.8/bakery/run-bakery && chmod +x ./run-bakery
$ sudo ./run-bakery init alpine-grub-efi
```

I then edited `layers/customized.toml` and in the _recipes_ array added `"core/ssh",` and at the bottom added:

```
[parameters."core/ssh"]
root_authorized_keys = """
<INSERT YOUR PUBLIC SSH KEY HERE>
"""
```

```
$ sudo ./run-bakery bake image customized-amd64
$ ls -lh build/customized-amd64/system.img 
-rw-r--r-- 1 root root 2.3G Apr 12 17:56 build/customized-amd64/system.img
```

koehlma, 2025-04-14:
The image includes redundant boot partitions and a config partition. That's why it is larger than you may expect. The actual system partition is not as large.

ben-auo, 2025-04-14:
I see, thank you for confirming.

closed 2025-04-14 by ben-auo

ben-auo, 2025-04-20:
zchunk might also help reduce size: https://www.thegoodpenguin.co.uk/blog/delta-ota-update-with-swupdate/

koehlma, 2025-04-20:
We already have something similar builtin, see: https://oss.silitics.com/rugix/docs/ctrl/over-the-air-updates#adaptive-delta-updates Also, this only reduces the size of update bundles, not of images.

ben-auo, 2025-04-20:
Thanks, I somehow missed reading about that feature. Any reason it's not the default? I'd be okay with a one-time large image setup and an infrequent. slow and large bundle creation, but the bandwidth costs for transferring the bundle to a large fleet of devices is prohibitive. The adaptive delta feature sounds perfect and the `install-update.sh` makes sense on the remote side but I'm a bit hazy on the server side.. What does the `$URL` point to -- the **system.rugixb** file?

koehlma, 2025-04-20:
It's a rather new feature (introduced in February). We will probably make it the default at some point with some additional logic to create the block indices on the fly. Yes, the URL simply points to the Rugix update bundle (`system.rugixb`). Rugix Ctrl uses HTTP range queries to retrieve individual parts of the bundle, which is supported by most HTTP servers out-of-the-box.

If I may ask, what are you building/working on?

ben-auo, 2025-04-21:
An Internet firewall appliance