rugix/rugix #67: Raspberry Pi 3 doesn't boot

reubenmiller, 2025-04-26:
Whilst upgrading the thin-edge.io integration with Rugix (migrating from Rugpi to Rugix), the u-boot based image does not seem to boot.

With the debug probe (using UART), the terminal shows that the u-boot console output showing that there is a problem loading the "second.scr" file.

```
U-Boot 2023.10 (Nov 22 2023 - 00:00:00 +0000)

DRAM:  948 MiB
RPI 3 Model B (0xa02082)
Core:  88 devices, 13 uclasses, devicetree: board
MMC:   mmc@7e202000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...
In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Net:   No ethernet found.
starting USB...
Bus usb@7e980000: USB DWC2
scanning bus usb@7e980000 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
** Booting bootflow 'mmc@7e202000.bootdev.part_1' with script
== Rugpi U-Boot First Stage ==
15 bytes read in 0 ms
17 bytes read in 1 ms (16.6 KiB/s)
Boot Spare:  0
Bootpart:  2
187 bytes read in 0 ms
Failed to load 'second.scr'
Executing second boot stage failed. Rebooting...
resetting ...
```

When switching back to the previous rugpi based image, the following shows the comparison to the u-boot output (which does boot):

```
U-Boot 2023.10 (Nov 22 2023 - 00:00:00 +0000)

DRAM:  948 MiB
RPI 3 Model B (0xa02082)
Core:  87 devices, 13 uclasses, devicetree: board
MMC:   mmc@7e202000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...
In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Net:   No ethernet found.
starting USB...
Bus usb@7e980000: USB DWC2
scanning bus usb@7e980000 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
** Booting bootflow 'mmc@7e202000.bootdev.part_1' with script
== Rugpi U-Boot First Stage ==
15 bytes read in 1 ms (14.6 KiB/s)
17 bytes read in 1 ms (16.6 KiB/s)
Boot Spare:  0
Bootpart:  2
187 bytes read in 1 ms (182.6 KiB/s)
502 bytes read in 0 ms
## Executing script at 01000000
== Rugpi U-Boot Second Stage ==
Bootpart:  2
Bootargs:  console=serial0,115200 console=tty1 rootfstype=ext4 fsck.repair=yes rootwait cgroup_memory=1 cgroup_enable=memory panic=60 root=PARTUUID=A3CD159B-05 init=/usr/bin/rugpi-ctrl
9276375 bytes read in 386 ms (22.9 MiB/s)
   Uncompressing Kernel Image
Moving Image from 0x80000 to 0x200000, end=1b60000
## Flattened Device Tree blob at 2eff7300
   Booting using the fdt blob at 0x2eff7300
Working FDT set to 2eff7300
   Using Device Tree in place at 000000002eff7300, end 000000002f002f2d
Working FDT set to 2eff7300
```

**Information**

|Name|Value|
|---|---|
|Rugix version|0.8.6|
|Repository|[thin-edge/tedge-rugpi-core@v0.8-rugix](https://github.com/thin-edge/tedge-rugpi-core/tree/v0.8-rugix)|

reubenmiller, 2025-04-26:
I wouldn't rule out some misconfiguration from my side (since there have been a lot of changes included along side the migration from Rugpi to Rugix)

reubenmiller, 2025-04-26:
Looking at the code where the second.scr file comes from, it seems that it is only included when I add the "core/rpi-uboot-setup" recipe...so it could be that I'm just not adding that in? Though I would of though that this would be automatically added in with the selection of the boot loader, otherwise it would make it a bit difficult to reuse the same image definitions, as you'd have to have one per boot loader (which wasn't previously required with rugpi). But maybe I'm just missing something...

reubenmiller, 2025-04-26:
Ok, when the `core/rpi-uboot-setup` recipe is added to the layer which should use u-boot, then the image is now bootable. However it is a little unexpected that it needs to be manually added given that target is set to `target = "rpi-uboot"`.

koehlma, 2025-04-27:
Thanks for the thorough investigation. Indeed, the `rpi-uboot-setup` recipe is necessary. The reason the second stage boot script is no longer added automatically when using the `rpi-uboot` target is that one might want to use a different script, e.g., to load specific device tree overlays. To build a layer that works for both `tryboot` and U-Boot, you can just add the recipe as it does not hurt to have the boot script there. This avoids having a layer per bootloader.

Will add that to the migration guide.

commit 41d566e0 (2025-04-27): docs: mention the necessity to add the `rpi-uboot-setup` recipe (fixes #67)
closed 2025-04-27 by koehlma