rugix/rugix #77: Bootstrapped GPT Custom Layout Example Complains about Root Path

SecareLupus, 2025-07-21:
I'm a little unclear from the documentation whether the system layout should be included in rugix-bakery.toml or installed to /etc/rugix/bootstrapping.toml via a recipe during the build process. I've seemingly gotten closer using rugix-bakery.toml's [systems.IMAGE_NAME.image.layout], using the example data structure from https://oss.silitics.com/rugix/docs/next/ctrl/bootstrapping/#custom-layouts

It's failing to complete the bake process though, complaining that
```
Error: partitions without a fixed size must have a root path
╰╴at crates/tools/rugix-bakery/src/oven/system.rs:460:29
```

Snippet from rugix-bakery.toml below:
```
[systems.rpi-stage4.image]
size = "32GiB"

[systems.rpi-stage4.image.layout]
type = "gpt"
partitions = [
    { name = "EFI", size = "256MiB", type = "C12A7328-F81F-11D2-BA4B-00A0C93EC93B" },
    { name = "boot-a", size = "256MiB" },
    { name = "boot-b", size = "256MiB" },
    { name = "system-a", size = "6GiB" },
    { name = "system-b", size = "6GiB" },
    { name = "data", filesystem = { type = "ext4" } },
]
```

I've tried adding a size of 16GiB to the data partition, but got the following error:
```
INFO Writing image partition table.
label: gpt
label-id: 60012985-BB74-4F55-87E2-D99AB43BBDD1
1: start=2048,size=524288,type=C12A7328-F81F-11D2-BA4B-00A0C93EC93B
2: start=526336,size=524288,type=0FC63DAF-8483-4772-8E79-3D69D8477DE4
3: start=1050624,size=524288,type=0FC63DAF-8483-4772-8E79-3D69D8477DE4
4: start=1574912,size=12582912,type=0FC63DAF-8483-4772-8E79-3D69D8477DE4
5: start=14157824,size=12582912,type=0FC63DAF-8483-4772-8E79-3D69D8477DE4
6: start=26740736,size=33554432,type=0FC63DAF-8483-4772-8E79-3D69D8477DE4

Error: unsupported GPT partition layout
╰╴at crates/tools/rugix-bakery/src/oven/system.rs:215:22
```

Am I using the wrong schema for the config file? I'm at a loss for how to get the image to expand properly (and complete boot) on firstboot.

koehlma, 2025-07-22:
The layout in `rugix-bakery.toml` is for the image used for initial device provisioning while the layout in `bootstrapping.toml` is applied at runtime during the bootstrapping process. This allows to build images without a second system and data partition. These partitions will then be created during the bootstrapping process. So, if you want a GPT layout for Raspberry Pi, you actually need both files with different configurations. You create a custom layout for the image which contains the EFI partition, both boot partitions, and the `system-a` partition. The layout in `bootstrapping.toml` has to mirror this layout and add the second system and data partition. You can use the layout from here as an example:

https://github.com/silitics/rugix/blob/682a248cc3db4cb306172b67f355ae65980838e3/crates/tools/rugix-bakery/src/oven/targets/mod.rs#L76-L120

You also need to root directories et cetera, otherwise the created filesystems will be empty.

This shows a GPT layout for Raspberry Pi (used with Yocto builds):

https://github.com/silitics/meta-rugix/blob/main/meta-rugix-rpi-tryboot/wic/sdimage-rugix-rpi-tryboot.wks.in

SecareLupus, 2025-07-23:
I feel like maybe I've overcomplicated the issue, as all I'm trying to do is bake two extra layers (and a handful of recipes) onto the raspi-os template. My guess is that the modifications I added got too big for the provided image's capacity. The status before playing with image layout or bootstrapping stuff was that I had a system.img file baked that wouldn't boot, because of something going wrong during the bootstrapping stage.

This is the error message I received (no manually configured image or bootstrap settings):
[image]

Which looks to me like the layout needs to be fixed, but maybe I'm way off base. I know the original template baked just fine without any manual configuration of the image layout, and seemed to have the A/B split automatically baked in. We ran early tests against it to evaluate the OTA process.

My SD card is more than large enough, at 64GiB, so that shouldn't be causing any external limitations, and the baked image I have is under 7GiB. Is there a better tack I should take on this? Maybe I'm overlooking something obvious?

koehlma, 2025-07-23:
I see, yeah, the default partition size for the system partitions is 4 GiB and your image is likely too large. You can just apply the [“default layout”](https://oss.silitics.com/rugix/docs/ctrl/bootstrapping#default-layout) with a different size by using a `bootstrapping.toml` configuration file with the following content:

```toml
[layout]
type = "default"
system-size = "8GiB"
```

This will give you 8 GiB system partitions. Please note that the size of system partitions cannot be changed after a system is provisioned (see the [production checklist](https://oss.silitics.com/rugix/docs/production-checklist#system-size)).

SecareLupus, 2025-07-23:
That worked for what I needed, thank you!

closed 2025-07-23 by SecareLupus